stm32duino/ST25DV #11: cannot read long text messages

ahmetcumhurarslan, 2021-07-10:
i could not read ndef messages longer than some limit. firstly the limit was shorter. i increased the "NDEF_TEXT_MAX_LENGTH" parameter and i could read longer messages. However i could not read more than 123 characters. can anybody give some hints to solve this issue? by the way i can read longer messages with my phone, so there is not an issue with the ic.

fpistm, 2021-07-12:
Hi @ahmetcumhurarslan 
Maybe you could try to change `NDEF_MAX_SIZE`:
https://github.com/stm32duino/ST25DV/blob/33ad94d8d8c287e215ed7a7965621ef8089e04e1/src/libNDEF/lib_NDEF.h#L61

I don't know this library so don't know the limitation of this component.

ahmetcumhurarslan, 2021-07-12:
i also changed this parameter but it did not worked.

fpistm, 2021-07-12:
OK. Would be too simple. 😉 
Could you give more details on your setup?
Core version? Arduino IDE version? Boards used? Options? Sketch used? Shield or components used? ...

ahmetcumhurarslan, 2021-07-12:
i am using Lolin esp8266 development board as microcontroller and "ant7-t-st25dv04k"  board as nfc sensor. i normally use platformio but for simplicity i made an example arduino sketch (attached as zip file). i tested it with arduino 1.8 ide.

[st25dv-test.zip](https://github.com/stm32duino/ST25DV/files/6804023/st25dv-test.zip)

ahmetcumhurarslan, 2021-07-12:
by the way i created "CustomST25DVSensor.h" file which replaces "ST25DVSensor.h". because it does not have a text reading function. the function is as follows:

```c++
int ST25DV::readText(String *text)
{
    uint16_t ret;
    NDEF_Text_info_t textInfo;
    sRecordInfo_t recordInfo;

    // increase buffer size for bigger messages
    ret = NDEF_ReadNDEF(NDEF_Buffer);
    if (ret) {
      return ret;
    }

    ret = NDEF_IdentifyBuffer(&recordInfo, NDEF_Buffer);
    if (ret) {
      return ret;
    }

    ret = NDEF_ReadText(&recordInfo,&textInfo);
    if (ret) {
      return ret;
    }

    *text = textInfo.text;

    return 0;
}
```

cparata, 2021-07-20:
Hi @ahmetcumhurarslan ,
did you try to increase beside NDEF_MAX_SIZE also the NDEF_RECORD_MAX_SIZE?
Best Regards,
Carlo

ahmetcumhurarslan, 2021-07-27:
i think i found the problem. Wire library has a buffer whose length is defined by "BUFFER_LENGTH". in our function we request long ndef messages at once so we cannot gather all of the characters. 

Simply changing below lines at function "ST25DV_IO_MemRead" of  "ST25DVSensor.cpp"

```c++
// be carefull with the (int)Size, the size parameter us no more on 16 bits but only 15 bits (the cast is required for arduino UNO)
st25dv._pwire->requestFrom((int)Addr, (int)Size); // Tell slave we need to read 1byte from the current register

while (st25dv._pwire->available()) {
  pData[i++] = st25dv._pwire->read();      // read that byte into 'slaveByte2' variable
}
```
 we can solve the problem as follows:

```c++
uint16_t totalSize = Size;
while(totalSize){
    uint16_t tempSize = totalSize>BUFFER_LENGTH ? BUFFER_LENGTH : totalSize;

    // be carefull with the (int)Size, the size parameter us no more on 16 bits but only 15 bits (the cast is required for arduino UNO)
    Wire.requestFrom((int)Addr, (int)tempSize); // Tell slave we need to read 1byte from the current register

    while (Wire.available()) {
        pData[i++] = Wire.read();      // read that byte into 'slaveByte2' variable
    }

    totalSize -= tempSize;
}

```

cparata, 2021-07-28:
Hi @ahmetcumhurarslan ,
thanks for the update. Did you also increase NDEF_MAX_SIZE and NDEF_RECORD_MAX_SIZE? Could you submit a Pull Request, so we can update the library with the fix?
Thanks in advance for your contribution and Best Regards,
Carlo

fpistm, 2021-08-09:
Well, with STM32 core, the `BUFFER_LENGTH `is the default size for Arduino compatibility but internally the buffer is allocated dynamically. So I guess you used this library with an other core? 
I guess we need this patch for Arduino compatibility.

cparata, 2021-09-17:
I'm waiting for a PR, so meantime we can close this issue for inactivity.

closed 2021-09-17 by cparata