stm32duino/ST25DV #12: Potentially Incorrect call of `Wire.endTransmission` after `Wire.requestFrom`

bsalinas, 2021-08-24:
I am using the ESP32 Arduino core and tracked down a bit of an obscure bug.
The examples weren't working for me (received errors saying that it wasn't possible to write or read).

Ultimately, I found that in paces where the library was calling `st25dv._pwire->requestFrom` followed by `st25dv._pwire->endTransmission`, the library was returning an error.
The ESP32 I2C implementation was returning error code 8 which is an ESP32 specific error code that signifies `I2C_ERROR_NO_BEGIN`. It is used when the user calls `endTransmission` without calling `beginTransmission`. That error code 8, results in `NFCTAG_ERROR` from `NFCTAG_ConvertStatus`

Looking at the Arduino examples for the Wire library, I think this could be the correct behavior. In those examples, when `Wire.requestFrom` (an I2C read) is called, `Wire.endTranmission` is not called. I think this indicates that `requestFrom` is responsible for sending the I2C stop condition for I2C reads. For I2C writes, I think `endTransmission` sends the stop condition.

Maybe the STM32Duino implements the Wire library in a different way (requiring you to call `endTransmission` after `requestFrom`). If so, there isn't going to be a simple solution to make this library work more broadly with different Arduino devices. 
I went ahead and made the issue and a fork (https://github.com/bsalinas/ST25DV) so that other people who face the same bug can have something to reference.

I'm happy to make a pull request, but I haven't tested the changes on the STM32duino, so I am worried about breaking it on other boards.

cparata, 2021-09-01:
Hi @bsalinas ,
thanks for the contribution. Your observation makes sense for me, so I suggest to submit the Pull Request. Then, I will test the patch with the STM32duino Core in order to check if everything is fine.
Thanks again for the help.
Best Regards,
Carlo

bsalinas, 2021-09-01:
Thanks, Carlo!

I do have some other changes I am making for configuring the GPO interrupt and for supporting other NDEF types. If you all would be interested in merging those in as well once they are more complete, let me know. It definitely adds some complexity to the library, so I totally understand if you want to keep the library more limited in scope.

cparata, 2021-09-01:
Hi @bsalinas ,
any improvement to the library is welcome. The only thing is that I prefer to have a different Pull Request for each new feature or bug fix in order to evaluate and integrate them one by one. Thanks in advance for your contribution.
Best Regards,
Carlo

cparata, 2021-09-17:
Fixed in the current master repo

closed 2021-09-17 by cparata