stm32duino/ST25DV #20: Can I change the UID of this tag?

spaniardmaximus, 2022-11-10:
Hi, I'm looking for a hardware and software combo allowing me to make rfid/nfc tags unique programmatically. So one device, multiple tags, in turn. Can this support it? Thanks beforehand.

cparata, 2022-11-21:
Hello @spaniardmaximus ,
what do you mean with UID? You can change the content of the tag as you want programming it by software.
Best Regards,
Carlo

spaniardmaximus, 2022-11-21:
Hi Carlo,

Thanks for your response. I refer to the UID as the unique id of the card/tag. The way that I understand it, you can add data to them, e.g. a url, as per example in README.md, no problem there. I've using an Arduino/pn532 library that can do that. 

What I'm really after is to "swap identities" with the same physical card/tag. Imagine having a list of ids mapped to users in an access control system. By changing the id on the same card/tag, I can present multiple users (in turn) to the same reader.

DISCLAIMER: Nothing nefarious in here :). Just trying to setup a testing rig with access control readers.

Cheers,

Israel.

spaniardmaximus, 2022-11-21:
Fair enough. Thanks.
Regards,
Israel.

closed 2022-11-21 by cparata