stm32duino/ST25DV #22: NDEF_WriteText function is missing language and encoding parameters

Gerriko, 2022-12-18:
The NDEF_WriteText function only allows for "en" language option via ISO_ENGLISH_CODE_STRING. There is no way to include other languages, like "de" for example.

It is also unclear whether the function has simply assumed it is UTF-8 encoding as I cannot see this specified in the code. Usually you have the option to specify either UTF-8 or UTF-16.

cparata, 2022-12-19:
Hello @Gerriko ,
you are right. At the moment these settings are hardcoded. We need to enhance the library to support also this features. I guess that the default settings is UTF-8 (encoding bit = 0). Any contribution is welcome to improve the library.
Best Regards,
Carlo

cparata, 2025-04-02:
Fixed in release 2.2.0