stm32duino/ST25DV #32: Callback function when the NFC chip is scanned

LauDijksterhuis, 2023-12-01:
Hi there,

I am new to electronics and only have a background in mobile app development. So to give you some context:
For my app I am building a 3D app icon in real life with a size of 10 x 10 x 2.5 cm which has some lights in it and a NFC chip (https://www.adafruit.com/product/4701) to redirect people to the AppStore using an https:// link.

These lights match the app icon as shown in the appstore and are controlled by a microcontroller. I would like to animate these the lights when a phone scans the NFC chip. Therefor I am looking for a "callback" function that gets triggered when the NFC chip gets scanned.

I have been looking in the code (I have no experience with C++ or NFC) but I could not find it.
Is this possible, or is there a way I can work on such a function to be implemented in this library?

cparata, 2023-12-01:
Hello @LauDijksterhuis ,
you can try to register an application callback on GPO pin of ST25DV component in this way:

```
volatile int gpo_event = 0;

void GPOEvent_cb()
{
  gpo_event = 1;
}

attachInterrupt(GPO_pin, GPOEvent_cb, RISING);
```

You can call the "attachInterrupt" in the setup function and then using the variable "gpo_event" in the loop function to understand when the smartphone is close to the tag and in this way you can drive your leds.
Let me know if it works on your side.
Best Regards,
Carlo

LauDijksterhuis, 2023-12-01:
Hi @cparata ,

Thank you so much!
I will try this weekend 😃

fpistm, 2024-01-11:
Closed as considered answered. @LauDijksterhuis feel free to give feedback if you have tested. Thanks