stm32duino/ST25DV #35: https://github.com/stm32duino/ST25DV/blob/2.1.2/src/libNDEF/lib_NDEF_Text.h#L38

dansiviter, 2025-03-21:
**Describe the bug**

[`NDEF_TEXT_MAX_LENGTH`](https://github.com/stm32duino/ST25DV/blob/2.1.2/src/libNDEF/lib_NDEF_Text.h#L38) set to an arbitrary and small value of 40. Text values can be set to larger that this using the ST23 Android app..

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Use ST25 Android app to create NDEF Text record larger that 40 characters,
2. Try and read data via I²C using `NDEF::NDEF_ReadText` method,
3. `NDEF_ERROR_MEMORY_INTERNAL` error returned.

**Expected behavior**
Ability to extract larger payloads either by default or be configurable.

**Desktop (please complete the following information):**
 - OS: N/A
 - Arduino IDE version: N/A
 - STM32 core version: N/A
 - Library version: 2.1.2
 - Tools menu settings if not the default: N/A
 - Upload method: N/A

**Hardware (please complete the following information):**
 - Board Name: Arduino Nano ESP32
 - Hardware Revision: 
 - Extra hardware used if any: 

**Additional context**
Add any other context about the problem here.

dansiviter, 2025-03-21:
For context, it seems this value is dynamic in other copies of the same library:[STMicroelectronics/STM32CubeL4#lib_NDEF_Text.h](https://github.com/STMicroelectronics/STM32CubeL4/blob/v1.18.1/Projects/B-L4S5I-IOT01A/Applications/NFC/Common/lib_nfc/lib_NDEF/Core/inc/lib_NDEF_Text.h#L44)

cparata, 2025-04-01:
Hello @dansiviter ,
thanks to have submitted this issue. Unfortunately, as the code is structured today, it is not easy to use a dynamic allocation for the NDEF buffer. What we can do easily it's to add the possibility to override the size of the NDEF buffer at application level. In this way, you can customize it according to your needs. We will try to add this feature in the next release of the library.
Best Regards,
Carlo

dansiviter, 2025-04-01:
Thanks @cparata , a customisable value would work for me.

fpistm, 2025-04-01:
Hi @dansiviter 
please, could you make the PR for this? 
Any contribution are welcome.

cparata, 2025-04-02:
Fixed in release 2.2.0

closed 2025-04-02 by cparata